dbaldwin/DronePan #3: Scrollable overview?

chrissearle, 2016-03-13:
Just a thought that struck me.

We have an issue with missing images - and to know if a particular sequence is OK or not you have to watch the tablet for each shot coming in.

If you threw each image (or a thumbnail) into the app's cache dir - you could add something like a collection view at the top of the page - one thumbnail high - scrollable left/right and add each image to that view as they come in.

Then the user could see as the sequence developed if there was a duplicate (no-movement) image.

You could clear the cache of images at the start of each sequence or not (depends if you want to add more functionality over "old" sequences or not I guess).

Should be a simple combo of UICollectionView, UICollectionViewCell and just storing images in cache as they come in (simple he says - I'm sure it's more complex than that - but I hope you get the idea?)

dbaldwin, 2016-03-14:
Very good idea. This would definitely be helpful. We are in the middle of a rewrite using SDK 3.0.1 and after that this is something we'll look into. This is the media API just as a reference:

https://developer.dji.com/iframe/mobile-sdk-doc/ios/Classes/DJIMedia.html

chrissearle, 2016-03-14:
Not sure I completely understand DJICompletionBlock - since it only seems to be passing an error if any - but - some combination of

    - (void)fetchThumbnailWithCompletion:(DJICompletionBlock)block

and the DJIMedia.thumbnail property should give you a UIImage (do you have to call the fetch to populate the property perhaps?). If you have a UICollectionView with UICollectionViewCells that can take the UIImage then you should be there.

You could also choose to add tap to see large etc - depends how far you want to go

chrissearle, 2016-04-17:
Tracking on https://github.com/chrissearle/DronePan/tree/feature/thumbnails

chrissearle, 2016-04-20:
The hard bit here is that in shoot mode it doesn't send the thumbnails. You have to switch to download mode. Options:

* toggle mode - adds time and complexity
* download at end - doable
* screengrab from videopreviewer - supported?