dbaldwin/DronePan #24: Gimbal Yaw for AC with yawable gimbal (Inspire)

chrissearle, 2016-04-20:
Enable in gui - not in beta 1 - we need AC yaw feedback.

chrissearle, 2016-05-30:
Is this a setting, a second start button, a popup dialog on start?

dbaldwin, 2016-05-30:
My thought is we make it a setting. The reason being is from numerous users I've talked to they are very opinionated about whether AC or gimbal yaw is better. Rarely do I hear that they use both. Well other than several suggestions that we use AC yaw for the sky row and gimbal for all the rest. But we can leave that for another day.

chrissearle, 2016-05-30:
One step at a time I think :) First round - setting.

chrissearle, 2016-06-14:
Setting in place - custom mission implemented - but - some issues to follow up with DJI

https://github.com/dbaldwin/DronePan/wiki/Custom-mission