dbaldwin/DronePan #38: Can we do a field stitch for pano preview?

dbaldwin, 2016-04-23:
It would be nice to do a preview stitch in the field as discussed here:

https://dji-dev.gitbooks.io/mobile-sdk-tutorials/content/en/iOS/PanoDemo/PanoDemo_en.html

This would allow pilots to see if there are any photo gaps after shooting.

bbbenji, 2017-01-20:
New link: 

https://developer.dji.com/mobile-sdk/documentation/ios-tutorials/PanoDemo.html

dbaldwin, 2017-01-20:
Thanks @bbbenji. Is this something you'd be willing to take a crack at?

Automatt, 2017-02-15:
I have an implementation of CVStitch using the c++ library that assembles 22 scaled-down pano images, but it's not fast -- it takes about 3 minutes on my 6s.  Results are okay for a preview.  I'm going to see if there's anything I can do to speed it up.

mjerris, 2017-02-15:
are you needing to do any preprocessing of the images such as resize prior to feeding it in?  if so check out libyuv.  it has a bunch of image resize and conversion functions that are well optimized for arm.

Automatt, 2017-02-15:
The resize is very fast - it's the stitching that takes a long time.  Right now I'm feeding all photos in at once.  I'm going to see if I get better results stitching them in batches, or one at a time.

dbaldwin, 2017-02-15:
@Automatt I would have to say that 3 minutes isn't too bad. Currently users don't have this option at all and in some small cases they shoot their photos, get to the office, and then realize there is a gap or duplicated photo. If I have to wait 3 minutes in the field to save me another trip out then to me it's a no-brainer. Let us know if you need any sample images.

BTW @GitZChen was working on a prototype a while back. He used a different lib but maybe I can get him to chime in and share his experience.

Automatt, 2017-02-23:
I made a prototype as a separate app.  I'm still working on it - it can't share the finished image yet.  But you can check a bunch of images in your photo library and it will stitch them together to make a low-fi preview.    Check it out here: [DroneStitch](http://github.com/Automatt/DroneStitch)

dbaldwin, 2017-02-24:
@Automatt that's killer! I'm going to give it a test this afternoon. Very exciting.