dbaldwin/DronePan #41: Support for AEB and HDR

dbaldwin, 2016-04-24:
It would be nice to support AEB and HDR photo modes. Here is some info as it relates to SDK 3.1, which is the version we're currently on:

**DJICameraShootPhotoModeHDR**
Sets the camera to take an HDR photo.
X5 camera, X5R camera and XT camera do not support HDR mode.

**DJICameraShootPhotoModeAEB**
Sets the camera to take a picture (or multiple pictures) continuously at a set time interval. The minimum interval for JPEG format of any quality is 2s. The minimum interval for Raw or Raw+JPEG format is 10s.

It seems the Inspire 1 X5 users will be out of luck with HDR for the time being, but perhaps we give them the AEB option.

chrissearle, 2016-04-26:
Once the settings are set - we need to check the camera events.

When trying for a shot - it will check for isShooting which we have defined as:

```
systemState.isShootingSinglePhoto ||
                systemState.isShootingSinglePhotoInRAWFormat ||
                systemState.isShootingBurstPhoto ||
                systemState.isShootingIntervalPhoto
```

So - that should wait until the burst or interval is done (assuming one of these two are involved - but again needs checking).

However - if a JPEG requires 2s per shot in AEB and RAW minimum 10s then our timeouts will give up.

We need therefore to detect if we're in this mode and extend timeouts.

TheSnoStorm, 2017-06-25:
In addition to being able to shoot AEB with a X5, what would seriously be nice would be the ability to shoot  either multiple AEBs, or 'custom AEB.' End goal of either solution would be to create high-range (EV +/-
 4-9) spherical HDRI images that are suitable to be used as environments/image based lighting [IBL] in 3D production.

Multiple AEB - The builtin AEB capability of the X5 is nice, but the broadest range it allows is 5 bracket +/- 0.7 . That only gives a dynamic range of -1.4 to 1.4 EV. So the idea would be that for every pano photo that needs taken it would first shoot an AEB at 0 EV, then adjust camera settings so that it was at -3.5 EV and shoot another AEB, then adjust camera settings so that it was at +3.5 EV and shoot another AEB. This will end up with 15 brackets ranging from -4.9 to +4.9.

Custom AEB - Basically take control of the camera and issue commands to create your own brackets. I know you can't modify the builtin AEB, so this would need to take single photos for each bracket. UI could be something like choosing the number of brackets you want as well as the step size between brackets.