dbaldwin/DronePan #42: Bug with pano not initializing in I1 and P4

dbaldwin, 2016-04-24:
This is a blocking bug:

https://www.facebook.com/groups/dronepan/permalink/255832421436013/

Let's figure out the root cause.

chrissearle, 2016-04-24:
Really hoping its something we can switch/toggle. Odd that it works for some but not others if it's not a setting or mode.

chrissearle, 2016-04-24:
Should we be checking isVirtualStickControlModeAvailable state of the FC?

chrissearle, 2016-04-25:
Possible workaround in https://github.com/dbaldwin/DronePan/commit/6c9786362ffaaf4c7dbb46daac995e01aa291e7e

chrissearle, 2016-04-25:
Comment from DJI:

"Our resetGimbal function only reset the Roll and Yaw for the Inspire 1. However, not for the pitch, we don’t rotate back to the 0 degree. We would add this detail to our documentation in the coming future version."

So - right now we work for I1 (and P4?) when the user is currently at pitch 0 but not otherwise.

The fix in https://github.com/dbaldwin/DronePan/commit/6c9786362ffaaf4c7dbb46daac995e01aa291e7e seems safe based on this.

chrissearle, 2016-04-25:
Fixed

closed 2016-04-25 by chrissearle