dbaldwin/DronePan #48: Check/set gimbal work mode?

chrissearle, 2016-04-26:
Check this log - http://l.facebook.com/l.php?u=http%3A%2F%2Fpastebin.com%2FPdSaXZpJ&h=GAQE5-jmm

We set a yaw but it never moves.

Wonder if we need to check gimbal work mode - possible values:

```
DJIGimbalWorkModeFreeMode
DJIGimbalWorkModeFpvMode
DJIGimbalWorkModeYawFollowMode
DJIGimbalWorkModeUnknown
```

We can set with ```gimbal setGimbalWorkMode:withCompletion:```

chrissearle, 2016-04-26:
Second report - http://pastebin.com/GBR35eBT

chrissearle, 2016-04-26:
Three i1s that won't reset because the yaw is not changing when we ask it to. But the users are saying that it is pointing towards craft forwards.

Is it possible that the gimbal yaw value is being sent relative to heading rather than relative to nose?

dbaldwin, 2016-04-26:
We should confirm that these users are on the latest firmware. The yaw value *should* be relative to the nose of the aircraft. From the SDK header file:

DJIGimbalState
"The current gimbal attitude in degrees. Roll, pitch and yaw are 0 if the gimbal is level with the aircraft and points in the forward direction of the aircraft."

chrissearle, 2016-04-26:
It's odd - the logs show very clearly that the yaw value is not changing at all.

chrissearle, 2016-04-26:
Feel free to ask in FB - I've asked - but not got a response yet. 

https://www.facebook.com/groups/dronepan/permalink/256940001325255/?comment_id=256942621324993&comment_tracking=%7B%22tn%22%3A%22R%22%7D

chrissearle, 2016-04-26:
DJIGimbalDelegate is here

https://github.com/dbaldwin/DronePan/blob/master/DronePan/GimbalController.swift#L316-L324

It's being called - but for some inspire's the yaw value is fixed - doesn't change.

chrissearle, 2016-04-28:
More reports from inspire users including i1 pro too now. They say that they can see the gimbal is pointing forwards but the value in the DJIGimbalState is fixed (and nearly always negative in each error report so far). It simply doesn't change.

This is only observed in 3.1 - haven't had a release to testers of 3.1.1 and not observable in 3.2 until #55 (DJIGimbalDelegate method not called) is fixed.

dbaldwin, 2016-04-29:
I changed the title to include include the P5. This commit fixes the delegate issue in SDK 3.2:

https://github.com/dbaldwin/DronePan/commit/185a901eeafaa6792309ddac55226b603babe235

and my P4 yaw value is not anywhere near zero even though the gimbal is pointed straight forward in the direction of the nose. See attached.
[image]

dbaldwin, 2016-04-29:
From DJI:

"I think for the Inspire 1 and Phantom 4, we now use the Ground System North as the 0 for the Yaw. ( Relative to the Nose). I think for our documentation about the GimbalState, we need to make some adjustment."

chrissearle, 2016-04-29:
I don't understand the DJI response.

* What is Ground System North relative to Nose - what does it mean?
* Where do we get the value for it?
* What value should we send in to point the gimbal towards the nose of the aircraft?
* What value should we check against in the gimbal delegate to see if it made it there?

chrissearle, 2016-05-06:
https://github.com/dbaldwin/DronePan/wiki/Yaw-Values

chrissearle, 2016-05-12:
So far it looks like:

* P3 and Osmo treat gimbal yaw 0 as relative to nose of craft
* P4 and I1 treat gimbal yaw 0 as relative to the current aircraft yaw

This is hard to test - since in the simulator - that gimbal yaw figure is relative to the physical aircraft yaw of the unit being tested as it sits on the bench - not relative to the simulated aircraft yaw (which is what we are seeing from the flight controller delegate.

Still not sure what the correct fix is here.

chrissearle, 2016-05-14:
I've added code to tell the gimbal controller what the aircraft yaw is.

Then - in check where we log what yaw is wanted and what the current yaw is - it also logs what the current AC yaw is.

We can then collect a set of logs - and compare - is it as simple as removing the ac yaw from current yaw for the P4 and I1?

mjerris, 2016-05-14:
Do you need some testing on this?  If so let me know what you need.  I have inspire, p4 and osmo

dbaldwin, 2016-05-15:
@mjerris when this beta goes to Testflight we'll definitely want help testing. Any and all testing is welcome since this issue has been plaguing us for a while.

commit 41cbebd1 (2016-05-16): Fixes #48. Relative gimbal yaw for P4 and I1
commit 1e417d37 (2016-05-16): Fixes #48. Relative gimbal yaw for P4 and I1
commit 9c22495f (2016-05-16): Fixes #48. Relative gimbal yaw for P4 and I1
commit 8e018e63 (2016-05-16): Fixes #48. Relative gimbal yaw for P4 and I1
closed 2016-05-26 by chrissearle