dbaldwin/DronePan #49: Product with no components

chrissearle, 2016-04-26:
This is a little worrying

```
I: 2016/04/26 10:10:16:290 [com.apple.main-thread/975699] ViewController -[ViewController sdkManagerProductDidChangeFrom:to:] line: 679 Change of product
I: 2016/04/26 10:10:16:290 [com.apple.main-thread/975699] ViewController -[ViewController sdkManagerProductDidChangeFrom:to:] line: 682 New product Inspire 1
E: 2016/04/26 10:10:16:291 [com.apple.main-thread/975699] ViewController -[ViewController sdkManagerProductDidChangeFrom:to:] line: 701 No FC found
E: 2016/04/26 10:10:16:293 [com.apple.main-thread/975699] ViewController -[ViewController sdkManagerProductDidChangeFrom:to:] line: 744 No camera found
E: 2016/04/26 10:10:16:294 [com.apple.main-thread/975699] ViewController -[ViewController sdkManagerProductDidChangeFrom:to:] line: 751 No gimbal found
```

chrissearle, 2016-04-26:
Seen with osmo too now in 3.1.1

We get a new product from the SDK - but - there is no camera or no gimbal or no battery or similar. Goes away if we restart the app.

chrissearle, 2016-04-27:
Reproduced in 3.2 - osmo seen but with no camera, gimbal or battery on resume from background. Was connected fine when it went into background.

chrissearle, 2016-04-28:
The product is returned from:

```- (void)sdkManagerProductDidChangeFrom:(DJIBaseProduct *_Nullable)oldProduct to:(DJIBaseProduct *_Nullable)newProduct```

but it sometimes has no camera, sometimes no gimbal, sometimes no battery and sometimes a combination.

dbaldwin, 2016-04-29:
From DJI:

Answer: - (void)sdkManagerProductDidChangeFrom:(DJIBaseProduct *_Nullable)oldProduct to:(DJIBaseProduct *_Nullable)newProduct

This method is to check whether the Product is connected. However, when talking about the component like camera, gimbal and battery, we use another method to check it in the DJIBasePorduct.h –(void)componentWithKey: (NSString *)key changeFrom: (DJIBaseComponent *_Nullable)oldComponent to : (DJIBaseComponent *_Nullable)newComponent;

chrissearle, 2016-04-29:
Believed fixed in d8c1ab5d27cf86c20ae545e222ef6c7defa57e60 - needs testing. Closing now - re-open if still an issue.

closed 2016-04-29 by chrissearle