dbaldwin/DronePan #50: DJIBatteryDelegate not called in 3.1.1

chrissearle, 2016-04-26:
Seems that after updating to 3.1.1 - the battery delegate (DJIBatteryDelegate) isn't seeing calls to battery didUpdateState - at least for Osmo.

chrissearle, 2016-04-26:
We set the delegate here

https://github.com/dbaldwin/DronePan/blob/master/DronePan/BatteryController.swift#L39

And the delegate method is

https://github.com/dbaldwin/DronePan/blob/master/DronePan/BatteryController.swift#L42-L48

I can step thru - it sets the delegate but never goes into didUpdateState.

Works in 3.1

dbaldwin, 2016-04-27:
Have you tested this on any aircraft yet?

dbaldwin, 2016-04-27:
Confirmed this is broken with the P4.

dbaldwin, 2016-04-29:
From DJI:

Answer: We tried both Phantom 4 and Osmo. The battery push info is normal. The following screenshot is the push info for the Osmo. Could you please provide us a piece of the code to let us figure out where the problem is.

chrissearle, 2016-04-29:
Fixed in d8c1ab5d27cf86c20ae545e222ef6c7defa57e60

closed 2016-04-29 by chrissearle