dbaldwin/DronePan #52: Incorrect model

chrissearle, 2016-04-26:
On moving to background and returning - you sometimes get the wrong model returned from the SDK.

Seen in 3.1.1 at least with Osmo - being identified as a Phantom 3 Professional.

dbaldwin, 2016-04-27:
I get Phantom 4 100% of the time during my testing. Will move on to others.

chrissearle, 2016-04-28:
OK - I have seen this several times on 3.1.1 - but not yet on 3.2 (but I haven't had a lot of testing time on 3.2 yet).

dbaldwin, 2016-04-29:
From DJI:

Answer: This seems to be a bug to us. We would record it and try to reproduce it.

chrissearle, 2016-04-29:
So far - still only repeated in 3.1.1 - could be very intermittent in 3.2 though

chrissearle, 2016-05-07:
Going to close this. Re-open if we see it in 3.2 or later

closed 2016-05-07 by chrissearle