dbaldwin/DronePan #54: Filenames for pano set

chrissearle, 2016-04-27:
We can't influence what goes on the card - but a suggestion came in to have a list of what shots are in a pano - and we can get the filenames for each shot taken.

Not sure of the GUI for this (people will need to see the info when they get home) - but - def. worth investigating.

dbaldwin, 2016-04-27:
It may simplify to display the start and final image vs all the images in between. Like DJI_0001.JPG - DJI_0020.JPG

Another suggestion that has come through multiple times is to separate panos with a 2s video clip.

lazyzero, 2016-04-27:
Is there the possibility to influence the exposure settings? Then you could read the settings, make a totally over/under exposed image and restore the old values.

chrissearle, 2016-05-06:
Include gimbal attitude and ac yaw for each too?

chrissearle, 2016-05-07:
Example log

```
I: 2016/05/07 16:30:18:181 [com.apple.main-thread/2156372] PanoramaController cameraControllerNewMedia line: 528 Shot taken: DJI_0127.jpg ACY: 0.0 GP: -60.0 GY: -0.4 GR: -0.6
I: 2016/05/07 16:30:25:713 [com.apple.main-thread/2156372] PanoramaController cameraControllerNewMedia line: 528 Shot taken: DJI_0128.jpg ACY: 0.0 GP: -30.0 GY: 0.3 GR: -0.2
```