dbaldwin/DronePan #65: LaunchScreen image

chrissearle, 2016-05-12:
We have a temporary LaunchScreen storyboard - but it needs replacing with something suitable.

chrissearle, 2016-05-14:
Discussion - which appoach:

* Splash screen - graphic - as per 1.3.7
* In accordance with Apple HIG

An extract from the https://developer.apple.com/library/ios/documentation/UserExperience/Conceptual/MobileHIG/LaunchImages.html

---

**Design a plain launch image that improves the user experience. In particular, the launch image isn’t an opportunity to provide:**

* An “app entry experience,” such as a splash screen
* An About window
* Branding elements, unless they are a static part of your app’s first screen

**Design a launch image that is identical to the first screen of the app, except for:**

* Text. The launch image is static, so any text you display in it won’t be localized.
* UI elements that might change. If you include elements that might look different when the app finishes launching, users can experience an unpleasant flash between the launch image and the first app screen.

dbaldwin, 2016-05-14:
Okay I'll take care of this, but could use your input on something. I'm used to defining launch images through the assets catalog given I'm going to use a static image. Or is it better to do this with the storyboard file. If the latter I need some guidance on the best way to achieve this. I'm going to use a basic screen like the one attached.
[image]

chrissearle, 2016-05-15:
The reason for using a storyboard is so that you can get closer to a 100% match between launch and main window independent of device. The behaviour of the simulator makes me think it creates a cached version for that device anyway (simulator > reset content and settings if you're seeing the iOS simulator not react to changes in the Launch storyboard).

To create the correct view - I think you'll need to do the following:

* Set the background color
* Add a menu view
* Add the button and text to the menu view
* Add the button to the main view and set the image to the start icon

Then copy the same constraints as you see in the main storyboard.

chrissearle, 2016-05-15:
We should probably change the grey view to black on the main view too.

dbaldwin, 2016-05-15:
I'll tackle it this evening. I realize it will take you 10 minutes to bang this out, but let me struggle through it so I can learn more about auto layout. Hopefully I'll have something for you to review by your am and then we can push a beta out.

chrissearle, 2016-05-15:
Sure - there's a reason it has your name on it under "Assignee" - a not so hidden "here's a chance to learn some storyboarding :)

dbaldwin, 2016-05-16:
I think I got this working but please give a pull and test. One thing I noticed is that for some reason we have the settings cog vertically offset from the super view by -2 pixels. I'll clean that up later but went head and applied the same to the launch screen storyboard. I'm sure this was my oversight at some point. I'll remove it in both main and launch storyboards tomorrow.

chrissearle, 2016-05-16:
Tested (made a tiny fix).

closed 2016-05-16 by chrissearle