dbaldwin/DronePan #71: Frozen battery status

EUGENELAPIN, 2016-05-20:
Hi!
In the beginning: my respect to you for DronePan! Hope to contribute a bit by testing solution.

We tried DronePan beta 1.4.3b8.
Flight set is:
- Inspire 1 Pro
- X5 + 45mm oly lens
- dual RC mode, iPad Air with DronePan was installed on the second RC
- alt 320m, windy 6-7m/s.

Program for Inspire was 20x9 +sky row.
First of all...we detected battery percentage freeze at 88%. it wasn't critical in our case - we monitored battery on pilot's RC, but in single RC mode - it can strain.

Current beta is not stable while switching apps (DJI Go <-> DronePan) and modes, we had to unload DronePan from memory and start again several times because there were no reaction on Play|Start button and picture translation was lost or frozen. It can be very untimely while shooting session is close to the limit of battery - we waisting time and battery to reload app.

I'm not sure but... it looks like yawing was by drone only, not by gimbal (as in the latest release). Or it should be set somewhere... We didn't find... It seems not so precise and we have breaks in 1-3 rows while another columns/shots are very close to each other. I'm not sure here, is it an algorythm depends on some settings? As an idea - forcing drone to yaw is very reasonable in "sky row" shooting to avoid props/legs in FOV. The rest process is enough for gimbal movement, am I right?

I hope there is a way to implement multibattery session (it is already realized in another specific purpose projects). Actually if I want gigapixel pano with narrow lens I have to spend time to find previous position/altitude and state azimuth manually to continue. I1Pro shoots about 120-140 pics per battery TB48 in an acceptable windy env, so all high quantity mods are unusable on it and at the same time unreasonable on Phantom with its wide FOV.

And one more idea: proper flying states to control flight time between drone maintenance, but it looks like this statistics in DJI App stops while switching to DronePan in single-RC mode (Phantom). Maybe it can be done transparently for DJI Go App or separately to keep in DronePan App... 

And thank you again for your work!

Eugene

chrissearle, 2016-05-20:
Ref battery - Can you get the logs (copy to clipboard button in settings window) ? If it's not showing battery updates then I'd hope we'd see why it doesn't have a battery object any more.

Ref switching apps - there's something fishy in how the SDK handles this - I even get the errors switching between DJI Go and the DJI testing bridge app - both written by DJI. We do want to find out how to make it more stable - but I'm not sure how yet. We're tracking this on #32

Ref gimbal yaw - this is disabled until we have a fix in place for #48 - DJI changed how the P4 and I1 report gimbal angle but not how the P3 models or Osmo do. Working on a fix.

Multiflight panos is further down the todo list (stability needs to be sorted first) - but would be related to similar changes from #9 - since the storing mechanism would be the same.

EUGENELAPIN, 2016-05-20:
Hi Chris!

Fist of all sorry for my inattention, I should read previous issues before (i.e. concerning gimbal ops disable). Glad to hear that many refs are already in work!

I've copied logs but... It looks like it starts with our second session on Phantom, no timeline from Inspire at all :( I had to copy it right after flight maybe... Is it cleaned every program start or every copy command? I will try to handle it carefully next time, but now no logs for battery issue :(. Let's keep it mind..

Thank you for the answer, we will continue with tests asap due to the weather conditions in region. Hope to be helpful :)

Best regards,
Eugene @ Sky.Work

chrissearle, 2016-05-20:
Right now it looks like we're set to roll log at a default size (when it reaches 1Mb) and also after 24 hours.

That time based one - I think we need to change that. Bit daft that you can't fly one day then send us the log the next - will probably extend that to a few days.

chrissearle, 2016-05-20:
I just pushed c40ad1dbb999453654b787f5d0bc619db335555e - sets it up for a maximum of 2Mb and 5 days.