dbaldwin/DronePan #72: Sky row naming

chrissearle, 2016-05-22:
Is the name "Sky Row" misleading? What we're saying is "allow it to go above (osmo -> below) the horizon. But - it's not a single row - if we allow above the horizon and also more than 3 rows (so 4 with the sky row) - we evenly space the rows - so more than one is above the horizon.

Actually - I'd like to change the behaviour so that it doesn't increase the row count - it's just a flag to say "start at horizontal" or "start at gimbal pitch limit".

chrissearle, 2016-05-23:
Thinking this should be a toggle:

* Upper limit is horizon (0˚)
* Upper limit is [gimbal limit in degrees]

(switch upper limit to lower limit for handheld).

Only issue is we'd need to feed the limit from the gimbal object into settings to show it.

Then number of rows is just "how many rows spaced evenly from the chosen limit.

dbaldwin, 2016-05-25:
I think that makes sense. Perhaps as additional clarification we can dynamically display the row spacing in degrees. For example:

- Upper limit = 30

- # rows = 6

- Row spacing = 20 degrees

chrissearle, 2016-05-26:
Related to #74

closed 2016-05-30 by chrissearle