dbaldwin/DronePan #101: Inspire 2 switch config

chrissearle, 2017-01-18:
Inspire RC switch - only has P, S, A - no F.

Needs adding to ModelConfig.plist

NosDE, 2017-02-16:
I think the Inspire 2 will not report the Value 3, when the FlightModeSwitch is in position "P".
It would be nice, when the DDLogDebug in the PanoramaController (L197) would print the RAW Value of remoteController.mode too. This could help to tackle down this Problem.

dbaldwin, 2017-02-16:
Have you tested with the latest public release and verified that it does not work? If you'd be willing to test an ad hoc build this would help us get a beta out for all I2 users. If you can hop into our gitter chat we can talk about getting you an ad hoc build (assuming the current version doesn't work for you). https://gitter.im/dbaldwin/DronePan

chrissearle, 2017-02-16:
All we know from the docs is the following:

```
public enum DJIRemoteControllerFlightModeSwitchPosition : UInt8 {
    
    /**
     *  Position One. For all products except Mavic Pro, this is the left most
     *  position of the flight mode switch on a remote controller from the
     *  perspective of the pilot. For example, on a Phantom 4 remote controller,
     *  Position One is labeled "A".
     *  For the Mavic Pro, this is the position that is furthest away from the
     *  pilot and labeled "Sport".
     */
    case One
    /**
     *  Position Two. For all products except Mavic Pro, this is the middle
     *  position of the flight mode switch on a remote controller from the
     *  perspective of the pilot. For example, on a Phantom 4 remote controller,
     *  Position Two is labeled "S".
     *  For the Mavic Pro, this is the position that is closest to the pilot
     *  (the P position).
     */
    case Two
    /**
     *  Position Three. For all products except Mavic Pro, this is the right
     *  most position of the flight mode switch on a remote controller from the
     *  perspective of the pilot. For example, on a Phantom 4 remote controller,
     *  Position Two is labeled "P".
     *  The Mavic Pro does not have a third position for the flight mode switch.
     */
    case Three
}
```

The change from reporting switch setting to position of 1, 2 or 3 and making every app maker have a table of what switch position is what for each model is such a poor decision. But we have to work with it.

That being said - I have no idea what switch position the i2 needs to be able to allow us to use flight controls from the SDK.

I went with 3 because it says it is the right most for all except the mavic and on all the other models the rightmost was the one we wanted.

If someone has an i2 then we need to test it with the different values.

NosDE, 2017-02-16:
I got a Log from Rik Bjil (he owns a I2)

D: 2017/02/14 13:57:08:009 [com.apple.main-thread/13351] PanoramaController checkRCMode() line: 197 Not in correct mode - saw DJIRemoteControllerFlightModeSwitchPosition for model Inspire 2

But in the Log there is no value...
So i recommend to change Line 197 of PanoramaController.swift to: