dbaldwin/DronePan #105: Phantom 4 Fails to Take Photo

phemmert, 2017-02-15:
Guys, I've been plagued with constant failures, using a Phantom 4. The hang usually happened after 3-5 images. Interestingly, when testing on the ground, the routine seems to work flawlessly. In the air? Not so well. I've appended the relevant log of one of my most recent failures. If you need more log data, please let me know. 

I: 2017/02/09 12:55:12:126 [com.apple.main-thread/26731] AppDelegate application(_:didFinishLaunchingWithOptions:) line: 70 DronePan launched
I: 2017/02/09 12:55:12:126 [com.apple.main-thread/26731] AppDelegate application(_:didFinishLaunchingWithOptions:) line: 72 Running version 1.6.4(21)
I: 2017/02/09 12:55:12:226 [com.apple.main-thread/26731] MainViewController viewWillAppear line: 96 Showing main window
D: 2017/02/09 12:55:13:742 [com.apple.main-thread/26731] ConnectionController sdkManagerDidRegisterAppWithError line: 85 Connecting to product
I: 2017/02/09 12:55:13:742 [com.apple.main-thread/26731] MainViewController sdkRegistered() line: 355 Registered
D: 2017/02/09 12:55:13:742 [com.apple.main-thread/26731] ConnectionController sdkManagerDidRegisterAppWithError line: 93 Connecting to real product
I: 2017/02/09 12:55:13:743 [com.apple.main-thread/26731] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIReachabilityChecker reachabilityFlagsChanged:] (GAIReachabilityChecker.m:161): Reachability flags update: 00000000
I: 2017/02/09 12:55:13:750 [com.apple.main-thread/26731] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher didSendHits:response:data:error:] (GAIBatchingDispatcher.m:223): Dispatch error: Error Domain=NSURLErrorDomain Code=-1009 "The Internet connection appears to be offline." UserInfo={NSUnderlyingError=0x1702578e0 {Error Domain=kCFErrorDomainCFNetwork Code=-1009 "The Internet connection appears to be offline." UserInfo={NSErrorFailingURLStringKey=https://ssl.google-analytics.com/batch, NSErrorFailingURLKey=https://ssl.google-analytics.com/batch, _kCFStreamErrorCodeKey=50, _kCFStreamErrorDomainKey=1, NSLocalizedDescription=The Internet connection appears to be offline.}}, NSErrorFailingURLStringKey=https://ssl.google-analytics.com/batch, NSErrorFailingURLKey=https://ssl.google-analytics.com/batch, _kCFStreamErrorDomainKey=1, _kCFStreamErrorCodeKey=50, NSLocalizedDescription=The Internet connection appears to be offline.}
I: 2017/02/09 12:55:13:852 [com.apple.main-thread/26731] ConnectionController sdkManagerProductDidChangeFrom(_:to:) line: 103 Connected to Optional("Phantom 4")
I: 2017/02/09 12:55:13:853 [com.apple.main-thread/26731] MainViewController connectedToProduct line: 365 New product Optional("Phantom 4")
D: 2017/02/09 12:55:13:853 [com.apple.main-thread/26731] PreviewController setMode line: 45 Trying to set hardware decoding
D: 2017/02/09 12:55:13:854 [com.apple.main-thread/26731] ConnectionController componentWithKey(_:changedFrom:to:) line: 184 New gimbal
I: 2017/02/09 12:55:13:857 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 67 Gimbal Controller init
D: 2017/02/09 12:55:13:857 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 118 Range extension supported: true
D: 2017/02/09 12:55:13:857 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 139 Gimbal Controller contraints pitch: A: true Mn: Optional(-90) Mx: Optional(30)
D: 2017/02/09 12:55:13:857 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 140 Gimbal Controller contraints yaw: A: true Mn: Optional(-15) Mx: Optional(15)
D: 2017/02/09 12:55:13:857 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 141 Gimbal Controller contraints roll: A: false Mn: nil Mx: nil
D: 2017/02/09 12:55:13:858 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlStiffnessYaw as range
D: 2017/02/09 12:55:13:858 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlGyroFilteringPitch as range
D: 2017/02/09 12:55:13:858 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 151 Logging AdvancedSettingsProfile as cap
D: 2017/02/09 12:55:13:858 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging ControllerDeadbandPitch as range
D: 2017/02/09 12:55:13:858 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging SmoothTrackAccelerationPitch as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlGyroFilteringRoll as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging EndpointPitchDown as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 151 Logging SmoothTrackEnabledYaw as cap
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlPrecontrolYaw as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlStrengthYaw as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlStrengthRoll as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging EndpointYawLeft as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging AdjustPitch as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging ControllerSpeedYaw as range
D: 2017/02/09 12:55:13:859 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging SmoothTrackDeadbandPitch as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging AdjustYaw as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging ControllerSpeedPitch as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 151 Logging SmoothTrackEnabledPitch as cap
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging SmoothTrackSpeedPitch as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging ControllerSmoothingYaw as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging ControllerDeadbandYaw as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging EndpointPitchUp as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlGyroFilteringYaw as range
D: 2017/02/09 12:55:13:860 [com.apple.main-thread/26731] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlPrecontrolRoll as range
[... cut ...]
D: 2017/02/09 12:56:01:370 [com.dronepan.queue/26753] ActiveAwareDispatchGroup leave() line: 22 Leaving dispatch group gimbal
D: 2017/02/09 12:56:01:370 [com.apple.root.default-qos/26773] ActiveAwareDispatchGroup wait() line: 55 Waiting complete for dispatch group gimbal
D: 2017/02/09 12:56:01:370 [com.apple.root.default-qos/26773] PanoramaController resetGimbal() line: 484 Reset gimbal - done
D: 2017/02/09 12:56:01:370 [com.apple.root.default-qos/26773] PanoramaController doPanoLoop() line: 457 PanoLoop: END

dbaldwin, 2017-02-16:
Thanks for sharing this. For some reason the camera is busy and the retries don't work. As a test will you set the camera delay to 5s in your settings and see what happens then? We're trying to work around this but seeing intermittent results. I realize a 5s delay between photos isn't ideal but please give this a test and let us know.

phemmert, 2017-02-16:
Will do, am traveling so will have to wait until this weekend. Hopefully we'll find a cure.

phemmert, 2017-03-02:
Ok, went out again today, set 5 second interval, resulting in the same failure symptoms. Oddly, prior to this, with 5 seconds, it worked fine. Today not so well. The problem is intermittent, which I'm sure makes for difficult troubleshooting. Today's logs follow:

I: 2017/03/02 12:59:27:123 [com.apple.main-thread/122092] AppDelegate application(_:didFinishLaunchingWithOptions:) line: 70 DronePan launched
I: 2017/03/02 12:59:27:124 [com.apple.main-thread/122092] AppDelegate application(_:didFinishLaunchingWithOptions:) line: 72 Running version 1.6.4(21)
I: 2017/03/02 12:59:27:227 [com.apple.main-thread/122092] MainViewController viewWillAppear line: 96 Showing main window
D: 2017/03/02 12:59:29:215 [com.apple.main-thread/122092] ConnectionController sdkManagerDidRegisterAppWithError line: 85 Connecting to product
I: 2017/03/02 12:59:29:215 [com.apple.main-thread/122092] MainViewController sdkRegistered() line: 355 Registered
D: 2017/03/02 12:59:29:215 [com.apple.main-thread/122092] ConnectionController sdkManagerDidRegisterAppWithError line: 93 Connecting to real product
I: 2017/03/02 12:59:29:215 [com.apple.main-thread/122092] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIReachabilityChecker reachabilityFlagsChanged:] (GAIReachabilityChecker.m:161): Reachability flags update: 0X000002
I: 2017/03/02 12:59:29:231 [com.apple.main-thread/122092] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher didSendHits:response:data:error:] (GAIBatchingDispatcher.m:226): Hit(s) dispatched: HTTP status 200
I: 2017/03/02 12:59:29:248 [com.apple.root.default-qos.overcommit/122122] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher deleteHits:] (GAIBatchingDispatcher.m:530): hit(s) Successfully deleted
I: 2017/03/02 12:59:29:259 [com.apple.root.default-qos.overcommit/122122] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher didSendHits:] (GAIBatchingDispatcher.m:237): 2 hit(s) sent
I: 2017/03/02 12:59:31:398 [com.apple.main-thread/122092] ConnectionController sdkManagerProductDidChangeFrom(_:to:) line: 103 Connected to Optional("Phantom 4")
I: 2017/03/02 12:59:31:402 [com.apple.main-thread/122092] MainViewController connectedToProduct line: 365 New product Optional("Phantom 4")
D: 2017/03/02 12:59:31:402 [com.apple.main-thread/122092] PreviewController setMode line: 45 Trying to set hardware decoding
D: 2017/03/02 12:59:31:404 [com.apple.main-thread/122092] ConnectionController componentWithKey(_:changedFrom:to:) line: 184 New gimbal
I: 2017/03/02 12:59:31:408 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 67 Gimbal Controller init
D: 2017/03/02 12:59:31:410 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 118 Range extension supported: true
D: 2017/03/02 12:59:31:410 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 139 Gimbal Controller contraints pitch: A: true Mn: Optional(-90) Mx: Optional(30)
D: 2017/03/02 12:59:31:410 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 140 Gimbal Controller contraints yaw: A: true Mn: Optional(-15) Mx: Optional(15)
D: 2017/03/02 12:59:31:410 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 141 Gimbal Controller contraints roll: A: false Mn: nil Mx: nil
D: 2017/03/02 12:59:31:411 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlStiffnessYaw as range
D: 2017/03/02 12:59:31:411 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlGyroFilteringPitch as range
D: 2017/03/02 12:59:31:412 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 151 Logging AdvancedSettingsProfile as cap
D: 2017/03/02 12:59:31:412 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging ControllerDeadbandPitch as range
D: 2017/03/02 12:59:31:412 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging SmoothTrackAccelerationPitch as range
D: 2017/03/02 12:59:31:412 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlGyroFilteringRoll as range
D: 2017/03/02 12:59:31:412 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging EndpointPitchDown as range
D: 2017/03/02 12:59:31:412 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 151 Logging SmoothTrackEnabledYaw as cap
D: 2017/03/02 12:59:31:413 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlPrecontrolYaw as range
D: 2017/03/02 12:59:31:413 [com.apple.main-thread/122092] GimbalController init(gimbal:gimbalYawIsRelativeToAircraft:allowsAboveHorizon:) line: 148 Logging MotorControlStrengthYaw as range
[... cut ...]
I: 2017/03/02 14:56:08:812 [com.apple.main-thread/129308] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher didSendHits:response:data:error:] (GAIBatchingDispatcher.m:226): Hit(s) dispatched: HTTP status -1
I: 2017/03/02 14:56:08:814 [com.apple.root.default-qos.overcommit/129334] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher deleteHits:] (GAIBatchingDispatcher.m:530): hit(s) Successfully deleted
I: 2017/03/02 14:56:08:818 [com.apple.root.default-qos.overcommit/129334] AnalyticsUtils info line: 83 GoogleAnalytics 3.17 -[GAIBatchingDispatcher didSendHits:] (GAIBatchingDispatcher.m:237): 1 hit(s) sent

amirfarazmand, 2017-07-27:
The only different in air and on the ground is possibly your communication is not as good up in the air. Try when the aircraft is only a few metres away in air.

phemmert, 2017-07-28:
This thread is outdated. The Phantom works fine now. The Inspire 2 is a different story, to be tackled another day.

amirfarazmand, 2017-07-28:
You also develop for Android?

dbaldwin, 2017-07-28:
@phemmert If you are testing on Inspire 2 please open a new issue and let us know what errors you're seeing. We'll be glad to take a look.