dbaldwin/DronePan #107: DJI Spark support?

imZack, 2017-07-17:
Mobile SDK 4.2 already supports spark, any plan on adding this one?

PS. this project is awesome.

dbaldwin, 2017-07-17:
Hey @imZack absolutely. We are currently beta testing. If you'd like to get on the beta please join our FB group:

https://www.facebook.com/groups/dronepan/

and we are working on adding waypoint missions to save panos to be repeated. Here is the new 2.0  version in Swift in case you're interested:

https://github.com/dbaldwin/dronepan-swift/

I see you're quite an active developer. Please let me know if you're ever interested in getting involved with DronePan dev.

Thanks for the comment.

imZack, 2017-07-18:
Good, I have already joined FB group and I should just check the Swift one for the latest beta version.
Yep, I'm very interested in DronePan dev and I just get my first drone which is DJI Spark.

closed 2017-07-18 by imZack

dbaldwin, 2017-07-18:
That's great. I'd love to get you involved so feel free to reach out to me with any questions. We have a big update coming soon and maybe you can get into the code after that. Thanks for reaching out.